ElderlyPirate/Bus_Pirate #126: Problem with $ command not working

Twilight-Logic, 2019-02-10:
There seems to be a problem with the $ command with some versions of the bootloader/firmware. Not sure  which is responsible.

My original v3 BusPirate board, version 3.6a came with firmware version 5.1. It was necessary for me to update the firmware to at least version 6.2. Since the firmware was updated, the $ command has not worked although it is still possible to update the firmware by using the hardware link.

Since upgrading, when issuing the $ command, the firmware responds with the word BOOTLOADER but does not respond to further prompts, i.e. it 'hangs'.

I have tried bootloader 4.4 and 4.5 with the same result. Firmware versions tried include:

BPv3-firmware-v6.2-r1981.hex
BPv3-firmware-v6.3-r2151.hex
S_1-05112018.hex (v7.0?)

I have not tested the experimental version U_1-05112018.hex (v7.x?)

USBEprom, 2019-02-10:
@Twilight-Logic 

Since firmware U_1-05112018.hex is the same as firmware S_1-05112018.hex except that this last one does not have the I2C hardware protocol unlocked, and you wrote it does not work while in UART transparent bridge and live monitor, so I think it would be very unlikely that it would work for you.
In any case it is up to you whether to try it or not.
The only other firmware v7.x ready to use I know is the BPv3-firmware-v7.1-1.hex in the path Bus_Pirate-master\package\BPv3-firmware\ inside master.zip.
You could even try that, if you did not already.
However, please note that for what I can verify with my Bus Pirate that for what I can understand is the same as to yours, the firmwares U_1-05112018.hex and S_1-05112018.hex with bootloader v4.5 have absolutely no the problem you wrote, as well as up to now nobody except you has never had the problem you wrote by updating to any firmware v7.x released, except in cases where it was necessary the new bootloader v4.5 that has fixed the problem.
Please take a look at these:

http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&p=67056#p67056

https://github.com/BusPirate/Bus_Pirate/issues/38

https://github.com/BusPirate/Bus_Pirate/issues/58

Could you please provide a picture of both sides of your Bus Pirate?
Thanks.

Could you please put here the answer to the "i" command of your Bus Pirate?
Thanks.
I know you already did it, but in another thread though, for the benefit of all it would be useful if you bring it also here.

Twilight-Logic, 2019-02-10:
I have re-tested. U_1-05112018.hex in the file I downloaded from your link in file BPv3-bootloaderv4xtov45-update-vb_ENG-h.7z does have the problem. However, having now tested U_1-05112018.hex in the same archive, I found that this does respond OK to the $ command but...

When I tried to upload a firmware, the upload failed on the first line of page 01:

> ./ds30 --dev=/dev/ttyUSB2 --hex=v6.2-r1981.hex
> +++++++++++++++++++++++++++++++++++++++++
> + DS30Loader client for Buspirate v2/v3 +
> +++++++++++++++++++++++++++++++++++++++++
> 
> Parsing HEX file [v6.2-r1981.hex]
> Found 21502 words (64506 bytes)
> Fixing bootloader/userprogram jumps
> Opening serial device /dev/ttyUSB2...OK
> Configuring serial port settings...OK
> Sending Hello to the Bootloader...OK
> 
> Device ID: PIC24FJ64GA002 [d4]
> Bootloader version: 1,02
> Erasing page 0, 0000...OK
> Writing page 0 row 0, 0000...OK
> Writing page 0 row 1, 0080...OK
> Writing page 0 row 2, 0100...OK
> Writing page 0 row 3, 0180...OK
> Writing page 0 row 4, 0200...OK
> Writing page 0 row 5, 0280...OK
> Writing page 0 row 6, 0300...OK
> Writing page 0 row 7, 0380...OK
> Erasing page 1, 0400...ERROR

Fortunately I was able to remedy this using the link.

So yes, the $ command seems to work in this firmware U_1-05112018.hex, but there seems to be an issue with the upload process when using that method.

USBEprom, 2019-02-11:
@Twilight-Logic

Thanks for having provided what was asked you and for your detailed reports.
So in the end current firmware v7.x luckily has not the problem, very well!
At this point I believe that the issue can be closed.
Thanks a lot.

Twilight-Logic, 2019-02-12:
Thanks for looking into it, but yes agree. This issue can now be closed.

closed 2019-02-12 by Twilight-Logic