ElderlyPirate/Bus_Pirate #128: Syntax error at char 2

ArtificialAmateur, 2019-04-27:
When I boot up the BPv4 and tap in with picocom this command is sent:
``ATE1 E0 ATE1 E0 ATE1 E0 ~x~~x~``

Which results in a self-test to start and to print out `Syntax error at char 2`

wiperpaul, 2019-10-25:
Same Issue here
thought it was random pattern but exact same as you.

wiperpaul, 2019-10-28:
for future reference these commands seem to be sent by modemmanager 
try: ``systemctl stop ModemManager``

hlovdal, 2021-05-09:
The more permanent fix to tell ModemManager to ignore the device is to create a custom udev rule, say `/etc/udev/rules.d/buspirate.rules`.

The following is to have MM ignore a mobile phone, but it is trivial to replace the ids with the ones used by buspirate.

```text
# http://linux-tips.org/t/prevent-modem-manager-to-capture-usb-serial-devices/284/2.

#$ lsusb | grep Sony
#Bus 007 Device 006: ID 0fce:d0d9 Sony Ericsson Mobile Communications AB 
#$

#udevadm control --reload-rules

ATTRS{idVendor}=="0fce" ATTRS{idProduct}=="d0d9", ENV{ID_MM_DEVICE_IGNORE}="1"

```