ElderlyPirate/Bus_Pirate #130: Bus Pirate resetting itself while performing script mode (aka BASIC).

USBEprom, 2019-05-14:
Based on what the user Lincon wrote on the forum of dangerousprototypes about an issue that concerns the LSb/MSb setting is not honored when sending data in script mode:

http://dangerousprototypes.com/forum/viewtopic.php?f=4&t=10260

I found that something is wrong with BASIC script mode because with the latest firmwares 09032019_OPT1_SAFE.hex and 09032019_OPT1_UNSAFE.hex (http://dangerousprototypes.com/forum/viewtopic.php?f=28&t=8498&start=120#p68332) based on the current repository the Bus Pirate resetting itself while running scripts, so that BASIC does not work anymore.

Thanks.

tudstudent, 2026-10-22:
Fixed by #212 - the board reset came from the 2018 store-copy rewrite (out-of-bounds write, corrupted program area, unbounded walk, address error trap, and the default trap handler is a reset). The same PR restores the expression parser from the same cleanup series, bounds every walk and index, and adds the missing BASIC documentation (Documentation/basic.md). Verified on hardware: 27 checks on the exact PR head, Bus Pirate v4, including the before/after of each failure: https://github.com/ElderlyPirate/Bus_Pirate/pull/212

USBEprom, 2026-10-24:
**@tudstudent**

Thank you Sir, respect!
Hoping all this will be merged in the code.

Be seeing you. 

U.Sb