ElderlyPirate/Bus_Pirate #131: flashrom support broken

ArtificialAmateur, 2019-05-19:
I've been having issue after issue of trying to get my BPv4 to properly flash dump.

v7.1 firmware: "Initializing buspirate_spi programmer"

6.2-r1981 / v7.0 / v6.3-r2151: Chip either isnt detected half the time or the other half it reads the entire binary as all "0xFF" or "0x00".

I've swapped out my FT2232H in between each time to assure it isnt the target chip or the cables.

USBEprom, 2019-05-19:
Hi.
Please, what exactly is the target chip you are dealing with?
What is the release of flashrom you are using?
What is the operating system you are running?
Thanks.

ArtificialAmateur, 2019-05-23:
>  Please, what exactly is the target chip you are dealing with?
Winbond W25Q80DV

>  What is the release of flashrom you are using?
flashrom-1.1-rc1

>  What is the operating system you are running?
Debian 4.19.37-3 (2019-05-15)

USBEprom, 2019-09-07:
Hi guys.
Today while I was doing some tests with some SPI memories I have sadly noticed that FLASHROM no longer works properly.
The latest working firmwares are version S_1-28102018.hex and U_1-28102018.hex which are located here:

http://dangerousprototypes.com/forum/index.php?topic=8498.msg67796#msg67796

All the firmwares released later do not work properly with FLASHROM probably due to some incompatibility introduced with the patch used to correct timing in the protocols 2-WIRE, 3-WIRE and 1-WIRE, as the culprit seems to be due to the management of delays:

https://github.com/BusPirate/Bus_Pirate/issues/23#issuecomment-435584095

Indeed, using the commands "&"  and "%" of the Bus Pirate directly from the terminal no delay is introduced as if they were ignored.
For its part FLASHROM in my case is able to read the chips but not to delete/write them.

http://dangerousprototypes.com/forum/index.php?topic=8498.msg70075#msg70075

Be seeing you.

U.Sb

ipatch, 2020-01-02:
I spent a little time messing around with this issue this evening, and can confirm the latest flashrom [v1.1](https://download.flashrom.org/releases/) does not work with the latest compiled version of this firmware.  I tried a different firmware from the dangerousprototypes forum, and could not get that to work either.

For the time being I downgraded flashrom from 1.1 to v1.0.1 and created a custom formula so flashrom would compile / build against the libftdi v1.4 library as opposed to the older v0.2 of the libftdi library.

And yes I tried flashrom with this firmware compiling against both the libftdi libraries and got the same results with both version of flashrom 1.1

I am able to successfully build the firmware from the master branch of this repo and use the mac pirate loader utility to load bootloader 4.3 and FW CE 7.1 and flashrom was able to dump the contents of a winbond flash chip.

All this was performed on macOS 10.13.6

For anyone else running into similar issues running macOS, I have a flashrom formula for homebrew [here](https://github.com/ipatch/homebrew-us-05)

robertodormepoco, 2020-01-17:
Hi, same problem here

i had to go back to v6.1 (BP v3.5) and @ipatch custom formula allowed me to make flashrom work (though my rom is not recognized and does not seem to be in list :( )

for the sake of trying, using flashrom 1.1 i get this (that has an updated list of supported devices)

? /usr/local/Cellar/flashrom/1.1/bin/flashrom -p buspirate_spi:dev=/dev/tty.usbserial-A500E4N4,spispeed=1M -r flash.bin  --verbose
flashrom v1.1 on Darwin 18.7.0 (x86_64)
flashrom is free software, get the source code at https://flashrom.org

flashrom was built with LLVM Clang 10.0.1 (clang-1001.0.46.4), little endian
Command line (5 args): /usr/local/Cellar/flashrom/1.1/bin/flashrom -p buspirate_spi:dev=/dev/tty.usbserial-A500E4N4,spispeed=1M -r flash.bin --verbose
Calibrating delay loop... OS timer resolution is 1 usecs, 2096M loops per second, 10 myus = 10 us, 100 myus = 115 us, 1000 myus = 1003 us, 10000 myus = 10020 us, 4 myus = 5 us, OK.
Initializing buspirate_spi programmer
Detected Bus Pirate hardware 3.5
Detected Bus Pirate firmware 6.1
Using SPI command set v2.
SPI speed is 1MHz
Bus Pirate v3 or newer detected. Set serial speed to 2M baud.
Warning: given baudrate 2000000 rounded down to 230400.

so I'd say that an old flashrom version/BP firmware is the way to go, sadly i can't confirm it is working 100% in my case due to the lack of supported device

ipatch, 2020-01-17:
hi @robertodormepoco 

this is the setup i'm currently using on my bus pirate. i have the, **BPv3a** from September 2009, and the software stack on my pirate is,

```
HiZ>i
Bus Pirate v3.a
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE PIC DIO] Bootloader v4.3
DEVID:0x0447 REVID:0x3003 (24FJ64GA00 2 A3)
http://dangerousprototypes.com
HiZ>
```

and i have flashrom pinned to **v1.01** on my macbook running macOS 10.13.6

```
flashrom v1.0.1 on Darwin 17.7.0 (x86_64)
flashrom is free software, get the source code at https://flashrom.org
```

i don't remember if i tested on any Linux based distros, but the above mentioned setup allowed me to read/dump the contents of the winbond chip on my open bench logic sniffer.

ipatch, 2020-05-20:
a little bump on this, I can no longer read an SPI flashchip using flashrom while mixing and matching various versions of flashrom, bp firmware, ftdi drivers on macos.

Skirmisher, 2020-10-11:
New BP user here (Sparkfun's v3.6a). Was working on flashing a Macronix chip, took a while to get it hooked up correctly (various silly issues) but flashrom would detect it. However, when trying to write to it, it would hang after detection before even getting to the initial read (i.e. before printing out "Reading old flash chip contents..."). I had installed BP firmware 7.1 (specifically [this tag](https://github.com/gdamjan/Bus_Pirate/releases/tag/v.test2) because it was convenient), and after reading this thread I went back and installed the last "official" firmware, 6.1. After doing that and retrying, flashrom no longer hung and could read/erase/write the chip correctly.

This was with flashrom version 1.2, libftdi version 1.4, on Fedora 32.

USBEprom, 2020-10-11:
@Skirmisher

About the Bus Pirate revision v3, from http://dangerousprototypes.com/forum/index.php?topic=8498.msg70075#msg70075

the latest firmwares that are working with flashrom are version **S_1-28102018.hex** and **U_1-28102018.hex** which are located here:

http://dangerousprototypes.com/forum/index.php?topic=8498.msg67796#msg67796

Direct download from here:

http://dangerousprototypes.com/forum/index.php?action=dlattach;topic=8498.0;attach=9896

------

From http://dangerousprototypes.com/forum/index.php?topic=8498.msg70165#msg70165

there are also **S_1-29092019.hex** and **U_1-29092019.hex** that are exactly the same **S_1-28102018.hex** and **U_1-28102018.hex** as before but rebuilt with the agatti fix for avrdude in spi.c. (http://dangerousprototypes.com/forum/index.php?topic=8498.msg70164#msg70164)

Direct download from here:

http://dangerousprototypes.com/forum/index.php?action=dlattach;topic=8498.0;attach=15167

------

And last, again from http://dangerousprototypes.com/forum/index.php?topic=8498.msg71171#msg71171

there are firmwares **busPirate-JTAG_SAFE_1.hex** and **busPirate_JTAG_UNSAFE_1.hex** that are exactly the same S_1-29092019.hex and U_1-29092019.hex as before with the JTAG fix from issue 134 (https://github.com/BusPirate/Bus_Pirate/issues/134) provided by Gabriel Smith as from his commit dated 21 May 2020.
Direct download here:

http://dangerousprototypes.com/forum/index.php?action=dlattach;topic=8498.0;attach=15705

------

You are right the old firmwares v6.x work with flashrom and also the firmwares v7.x up to the releases written in the opening.

wakass, 2021-06-19:
> I tried to look a little bit into the `flashrom` code, comparing the setup of the Bus Pirate in it with a hand made script and I found that with the following change in the file `buspirate_spi.c` (function `buspirate_spi_init()`)
> 
> ```
> 	/* Set SPI config: output type, idle, clock edge, sample */
> -	bp_commbuf[0] = 0x80 | 0xa;
> +	bp_commbuf[0] = 0x80 | 0x8;
> ```
> 
> a flash device is found. It's configuration related with the `CKE` configuration parameter.
> 
> BTW it seems strange that such code, that appears to be pretty "old", haven't caused problem before.

Worked for me as well. 👍

rub3rth, 2022-02-04:
I'll bump this if that's allright
I have the same problem but no fixes mentioned in this thread works, does anybody have an idea on where this issue stands today? I'm on a 3.5 board and the "S_1-28102018.hex " firmware linked in this thread and I can't get flashrom to read any chips.
I'm trying with a couple, all of them listed as supported in flashrom. I've tried with virtually every firmware version I've found and all flashrom versions available both with and without the change in buspirate_spi.c mentioned by @gipi but nothing works (except for version 1.2, for some reason it doesn't compile but throws an error regarding const to enum conversion). Has anybody gotten this to work lately?

AreYouLoco, 2022-02-04:
I had the same problems with flashrom. I used buspirate just yesterday. I ended up with I think version 6.3beta that was the last one that worked with flashrom. When I am back from work I could send it to you. But I think its in repo.

On February 4, 2022 9:01:46 AM UTC, rub3rth ***@***.***> wrote:
> I'll bump this if that's allright
> I have the same problem but no fixes mentioned in this thread works, does anybody have an idea on where this issue stands today? I'm on a 3.5 board and the [...]

rub3rth, 2022-02-04:
Thanks for answering, I'll do another check with that firmware I guess. Did you use the latest version of flashrom, with or without any changes to the source?

AreYouLoco, 2022-02-04:
I use flashrom from debian bookworm repository with no modifications to the package. Not sure which version is it.

On February 4, 2022 9:48:00 AM UTC, rub3rth ***@***.***> wrote:
>Thanks for answering, I'll do another check with that firmware I guess. Did you use the latest version of flashrom, with or without any changes to the source? 
>

rub3rth, 2022-02-04:
Allright I understand, well, I guess the only difference left would be version of the board? I have a v3.5, now loaded with 6.3beta and using the same flashrom - no device found still! Thanks for you help though, I'm all out of ideas

VraiHack, 2023-09-02:
@gipi i wold like to re-open this topic guys, i have similar issue like.
I am capable to discover the SPI flash (ST M26P16) manually:
[image]

but not with flashrom:
[image]

I am using kali linux:
[image]

And i am using bus-pirate:
[image]

any suggestion please ? also where can i found the : flashrom code, that you mentioned about
Thank you

gipi, 2023-09-02:
@VraiHack the code is [here](https://github.com/flashrom/flashrom/blob/master/buspirate_spi.c#L693)

VraiHack, 2023-09-03:
Hi @gipi since i am using Kali linux the flashrom already installed, so what i did:
1-  i installed the [package ](https://github.com/flashrom/flashrom/tree/master)from github
2- then i made it the change in buspirate_spi.c
	/* Set SPI config: output type, idle, clock edge, sample */
-	bp_commbuf[0] = 0x80 | 0xa;
+	bp_commbuf[0] = 0x80 | 0x8;
3- I compile with make

**Result still the same:**
[image]

tudstudent, 2026-10-24:
Finding on the "Hangs on Initializing buspirate_spi programmer" symptom, from re-testing the protocol against current firmware:

I replayed flashrom's !buspirate_spi.c init sequence byte for byte against a Bus Pirate v4 running the current code (v8.0 lineage), and the complete initialization passes:

- `0x00` x20 -> `BBIO1`; `0x0F` -> banner parses as hardware `v4` / firmware `v8.0`, landing at `HiZ>`
- the hardware >= 3.0 baud block (`b`, `10`, divisor, space) completes and returns to `HiZ>`
- `0x01` -> `SPI1`; the four setup commands (`0x4B`, `0x67`, `0x8A`, `0x03`) all acknowledge
- both probe framings answer correctly: v2 (`0x04` write-then-read) replies ack + data; v1 (`0x02` / `0x10|n-1` exchange / `0x03`) replies acks + exchanged bytes

22/22 steps. So the hang does not reproduce against the current firmware - it belongs to the older releases named in the report.

The other half of the report (detection flaky, reads coming back all-FF/all-00) is separate: that depends on the chip, wiring and the MISO state, and is not covered by this finding. If anyone can still reproduce a hang on a current build, the useful capture is which of the steps above stops answering.

USBEprom, 2026-10-24:
**@tudstudent**

Thank you Sir, respect!
Hoping all this will be merged in the code.

Be seeing you. 

U.Sb

tudstudent, 2026-10-24:
Closing the remaining piece from the earlier analysis: a real flashrom run against a real target now passes end to end.

Setup: flashrom built from source for Windows (`v2.0.0-devel`, GCC 16.2.0 via MinGW-w64, `buspirate_spi` programmer only), driving a Bus Pirate v4 running this fork's firmware (build `dfb116e`), talking to a bus-level W25Q40-style SPI flash emulator (JEDEC `EF 40 13`).

Command and result:

```
flashrom -p buspirate_spi:dev=COM6,spispeed=30K -r read.bin
Found Winbond flash chip "W25Q40.V" (512 kB, SPI) on buspirate_spi.
...
Reading flash... done.
```

- The read-back image is byte-exact: all 524288/524288 bytes match the emulator's deterministic pattern (`addr ^ addr>>8 ^ addr>>16 ^ 0xA5`). The emulator logged 524544 bytes served.
- The SPI speed is pinned to 30 kHz on purpose: the emulator is a bit-banged slave with a clean decode ceiling around 50 kHz, and flashrom's default (8 MHz) is far beyond what it can decode.

So the original "Hangs on: Initializing buspirate_spi programmer" does not reproduce on the current tree: the complete `buspirate_spi` init sequence (BBIO entry, banner, baud block, SPI setup) and a full real read both complete. The init robustness from any terminal state comes from the simulated-reset fix carried in one of the open PRs (#225).