ElderlyPirate/Bus_Pirate #134: Typo in jtag.c

kevinhikaruevans, 2019-06-23:
Was just reading over jtag.c and noticed this:

```
      uint16_t words = 0;
      while (jtag_read_bit() == LOW) {
        words++;
        if (words < 250) {
          break; // 250 device timout/limit...
        }
      }

```

Should that be `words > 250`? It will `break` on the first loop iteration.

https://github.com/BusPirate/Bus_Pirate/blob/b580ca9dcda14b7c64b6be2ed1b4b376366b1738/Firmware/jtag.c#L103

kallisti5, 2019-06-25:
err.. yeah. That looks like a bug.  It was likely a while (words < 250) at some point.

ironiridis, 2020-06-14:
Fixed in dd7fbb0?

MrAureliusR, 2020-07-25:
I'm amazed this wasn't caught sooner. Wouldn't this have totally broken JTAG functionality?

USBEprom, 2020-09-20:
@yodaldevoid @ironiridis @everyone

Starting from https://github.com/BusPirate/Bus_Pirate/archive/dd7fbb0fedd27c08b9c33501ebbe4b28d8085cba.zip provided by **Gabriel Smith** aka **yodaldevoid**, I have built test firmware **busPirate_JTAG_UNSAFE_1.hex** which has hardware I2C freed and **busPirate-JTAG_SAFE_1.hex** which has not.
The two firmwares can be found here:

http://dangerousprototypes.com/forum/index.php?topic=8498.msg71171#msg71171

About the **SAFE** and **UNSAFE** version, please read these:

http://dangerousprototypes.com/forum/index.php?topic=8760.msg70072#msg70072
http://dangerousprototypes.com/blog/2009/08/10/bus-pirate-hardware-i2c-added/
http://dangerousprototypes.com/blog/2009/08/07/find-your-bus-pirates-pic-revision/

**Sadly at this moment I can not verify the functioning of the JTAG part, so I ask the courtesy to anyone who is able of doing it to try one of the two firmware and give a response, thanks.**

akhepcat, 2020-09-22:
Are there supposed to be firmware files  named  "busPirate_JTAG_UNSAFE_1.hex"  or   "busPirate-JTAG_SAFE_1.hex"

$ unzip -v dd7fbb0fedd27c08b9c33501ebbe4b28d8085cba.zip  | grep -i safe_ | wc -l
0

I'm otherwise not sure which firmware files you're referring to.

Also,  is this for BPv3  or BPv4?    - I have a BPv4 that I can test JTAG  with,  but only a BPv4  (currently community v7.0)

akhepcat, 2020-09-22:
Hopefully this leads to a  BPv4   fix, because I'd love to have JTAG  on my v4....

USBEprom, 2020-09-22:
@akhepcat ; @Leif

**busPirate_JTAG_UNSAFE_1.hex** and **busPirate-JTAG_SAFE_1.hex** are both firmware for Bus Pirate v3 (BPv3).
The two firmwares can be found here:

http://dangerousprototypes.com/forum/index.php?topic=8498.msg71171#msg71171

Sadly I can not verify them for the functioning of the JTAG part, therefore it is need that someone verify and report if they work or not.
Thanks.

USBEprom, 2020-09-26:
@gdamjan 

At first I did not understand the meaning of your message, but now I understand it and I must tell you that you have done an excellent job providing the community with a great service, you really had an excellent idea!
Thank you very much Sir!

USBEprom, 2020-09-26:
@akhepcat ; @leif

https://github.com/BusPirate/Bus_Pirate/pull/152

https://github.com/gdamjan/Bus_Pirate/releases/tag/v.test

akhepcat, 2020-09-26:
@USBEprom  -
HiZ>m
1. HiZ
2. 1-WIRE
3. UART
4. I2C
5. SPI
6. 2WIRE
7. 3WIRE
8. KEYB
9. PIC
10. DIO
x. exit(without change)

(1)>no mode change
HiZ>
HiZ>i
Bus Pirate v4
Community Firmware v7.1 - goo.gl/gCzQnW [HiZ 1-WIRE UART I2C SPI 2WIRE 3WIRE KEYB PIC DIO]
DEVID:0x1019 REVID:0x0003 (24FJ256GB106 A5)
http://dangerousprototypes.com

I don't see the JTAG enabled here, though everything else seems to be working fine.

USBEprom, 2020-09-26:
@Leif ; @akhepcat

Thank you for your reply.
As I wrote, sadly I do not have a Bus Pirate v4 for testing.

akhepcat, 2020-09-26:
@USBEprom   Yes - I do, so i'm testing as I can.   If you enable it, I can test it.  I just can't build it

USBEprom, 2020-09-27:
@leif ; @akhepcat @ALL

Although I do not have a Bus Pirate v4 to test it, I have built a firmware for that device where the JTAG option is enabled.
I do not know if it is working or not, I provide it only for testing purpose.
The package contains the **configuration.h** file used to compile the firmware, whose **optimization level used to build it is option 1**.
Please let me know if there is any problem or if it is need to make specific changes to the **configuration.h** file.
Thanks!

[FW_BPv4_JTAG_opt1.zip](https://github.com/BusPirate/Bus_Pirate/files/5288135/FW_BPv4_JTAG_opt1.zip)

gdamjan, 2020-09-27:
> @akhepcat ; @leif
> 
> #152
> 
> https://github.com/gdamjan/Bus_Pirate/releases/tag/v.test

those builds there don't have the jtag patch included (yet). Let me work on finishing up #152 and I can see about the other stuff later.

akhepcat, 2020-09-28:
@USBEprom  ( @gdamjan )   Just for the FWIW:
Flashed the   "opt1"  firmware on my v4,  but still no JTAG in the menu.

Looks like i'll be waiting for the jtag patches after #152 is complete.

Thanks.

USBEprom, 2020-09-28:
@leif ; @akhepcat

As I have already written, I do not own a Bus Pirate v4, I only have a Bus Pirate v3.
In any case I do not think that the JTAG item should appear in the menu, simply JTAG support can be active or not depending on what is set during compilation through the file **config.h**.
Firmware **FW_BPv4_JTAG_opt1.hex** that I provided here definitely has JTAG support enabled, as it is also possible to see from its **config.h** file attached.
Even the firmwares for the Bus Pirate v3 that I shared on dangerousprototypes.com forum have JTAG support enabled, but as far as I can see the wording JTAG does not appear in the menu.
Please, take a look at these:

http://dangerousprototypes.com/docs/JTAG

**NOTE: As of v5.9 the JTAG terminal library was moved to the Bus Pirate bonus/extras firmware.**

**Bus Pirate firmware v5+ (main firmware) includes support for the OpenOCD JTAG debugger instead, this is a much better option.**

http://dangerousprototypes.com/forum/index.php?topic=2207.0

**The ONLY THING removed from the firmware is JTAG terminal mode. NOBODY used JTAG terminal mode, and it will NOT help you to flash a device. It is not a console concept either. It is a way to entering raw byte values into JTAG registers. If you learned the flashing protocol (by hand), you could type in the raw byte values (100,000s of them) into the terminal mode, but nobody would do this, applications like OpenOCD automate it.**

Talking about the Bus Pirate v3, as of v5.9 the JTAG terminal library has been removed and moved to the Bus Pirate bonus/extras firmware.
I am not sure how things stand for Bus Pirate v4, though.
Bus Pirate v3 firmware v5+ (main firmware) includes support for the OpenOCD JTAG debugger instead, this is a much better option. 
I believe that the same thing, excluding numbering of the firmware, also applies to the Bus Pirate v4, so no JTAG item does exist in the menu.
But I repeat that I have never had or used a Bus Pirate v4, so I am not an expert on the subject.

About patch, the firmware **FW_BPv4_JTAG_opt1.hex** I provided here and the ones for Bus Pirate v3 named **busPirate-JTAG_SAFE_1.hex** and **busPirate_JTAG_UNSAFE_1.hex** that I have shared on dangerousprototypes.com forum (http://dangerousprototypes.com/forum/index.php?topic=8498.msg71171#msg71171), all them have the one in **jtag.c** provided from **Gabriel Smith** aka **yodaldevoid**, as follows:

`@@ -102,7 +102,7 @@ void jtag(void) {'

     'uint16_t words = 0;'
     'while (jtag_read_bit() == LOW) {'
       'words++;'
       'if (words > 250) {'
         'break; // 250 device timout/limit...'
       '}'
     '}`

I do not know if that is enough, though, because at this moment I can not verify the functioning of the JTAG side.

akhepcat, 2020-09-29:
Ah!   I missed where they took the jtag console option out, and I was using that as the indicator for support.    Mea culpa!

I hooked up my bp4  to my rPi, configured a local config file appropriately, and was able to get OpenOCD  to talk correctly to the bp4.

I did notice that sometimes when closing, it didn't reset the configuration correctly, so I had to button-reset the bp4  before re-running OpenOCD - that's not a terrible workaround, though.    The 'mode'  LED  lights up when things are working correctly, it seems.   

i haven't hooked it up to an actual JTAG host yet, since i think i killed my pogoplug fully and I don't have anything else laying around JTAG capable and "disposable"

But so far it's looking pretty good.

USBEprom, 2020-09-29:
@akhepcat

Thank you very much for sharing your experience, really much appreciated, thanks!
About **OpenOCD** with the **Bus Pirate v3** I have never encountered any problems in communication, not even with firmwares that are not too recent, but I never got past this, so I think this is not enough to determine if the problem has now been solved or not.
Honestly I did not understand what the problem that generated this issue consists of, so I do not understand how to replicate it and possibly check if it still exists or not.

tudstudent, 2026-10-22:
This is already fixed in the current tree: Firmware/jtag.c now reads 'if (words > 250)' in the read-ID loop, so the timeout works as intended. The issue can be closed.