ElderlyPirate/Bus_Pirate #135: Checksum does not match, line 6

T365, 2019-07-28:
Loading new firmware to my Bus Pirate 3.6A  I got this message:

+++++++++++++++++++++++++++++++++++++++++++
  Pirate-Loader for BP with Bootloader v4+
  Loader version: 1.0.2  OS: WINDOWS
+++++++++++++++++++++++++++++++++++++++++++

Parsing HEX file [C:\ONTWIKKELING\Arduino\Buspirate\Firmware_29-07-2019\busPirate.X.production.hex]
Checksum does not match, line 6
Could not load HEX file, result=-1

In MPLAB:
BUILD SUCCESSFUL (total time: 13s)
Loading code from O:/Arduino/Buspirate/Bus_Pirate-master/Bus_Pirate-master/Firmware/busPirate.X/dist/BusPirate_v3/production/busPirate.X.production.hex...
Loading completed

So no succes...

T365, 2019-08-02:
I tried to use the hex file in different tool now it's solved...

mike-pittelko, 2019-09-09:
load the file, convert to uppercase then save.  Use that and the checksum will be ok.

edesd, 2021-10-08:
> load the file, convert to uppercase then save. Use that and the checksum will be ok.

how to convert to uppercase??

jtcmedia, 2022-09-01:
> load the file, convert to uppercase then save. Use that and the checksum will be ok.

Did that, works until:

```
...
Writing page 41 row 332, a600...OK
Writing page 41 row 333, a680...OK
Writing page 41 row 334, a700...OK
Writing page 41 row 335, a780...OK
Erasing page 42, a800...ERROR [50]

Error updating firmware :(
```

Top-Dog, 2023-01-29:
This comment didn't make a lot of sense, so here's a bit more detail:

> load the file, convert to uppercase then save. Use that and the checksum will be ok.

The _busPirate.X.production.hex_ file generated by MPLAB has lowercase hex characters and the loader expects uppercase hex characters to pass the checksum. The easiest way to fix this is to open the generated hex file in Notepad++, select all, right-click and select UPPERCASE, and save over the old file.

> ```
> ...
> Writing page 41 row 332, a600...OK
> Writing page 41 row 333, a680...OK
> Writing page 41 row 334, a700...OK
> Writing page 41 row 335, a780...OK
> Erasing page 42, a800...ERROR [50]
> 
> Error updating firmware :(
> ```

This is a known issue (the firmware was successfully loaded) - it happens when the loader tries to program over the protected bootloader space, but since we're not changing the bootloader the loader just bails. See https://github.com/BusPirate/Bus_Pirate/issues/26#issuecomment-257071647 

The issue can probably be closed.

Herz3h, 2024-01-06:
thank you very much @Top-Dog