ElderlyPirate/Bus_Pirate #137: What hardware features would you like?

ddavidebor, 2020-04-06:
Hello beautiful people.

I work for a big ECAD software house, writing articles.
I would like to make a couple of articles improving the bus pirate V4 hardware.

Which features / improvements / bugfixes would you like ? only hardware-wise

David

mofosyne, 2020-04-08:
USB type-c connector at least? 

NFC tag for easy config (e.g. instant uart logging)?

Convenience function of linking the VPU (Voltage input for on-board pull-up resistors) to either the 3.3v or 5v output?

Enough cpu power to host an embedded webpage instead of just a serial console by pretending to be a usb network gadget with a local mdns url (e.g. buspirate.local )? (Thought this may be a bit of a wishful thinking).

ddavidebor, 2020-04-08:
> Okay I'll bite (In order of cost/absurdness):
> 
> * USB type-c connector at least?  (Allows for hooking up to your smartphone too!)
Definitely!
> * Convenience function of linking the VPU (Voltage input for on-board pull-up resistors) to either the 3.3v or 5v output?
could you elaborate a little bit on how this feature would be used?
> * NFC tag for easy config or status display (e.g. instant uart logging) (Especially if paired with a community run buspirate config script exchange app)?
I am actually writing about NFC at this very moment, and I like it, but I am not a fan of multiple ways of accessing a device. 
> * Variable voltage output from 0 to 5v ?
Definitely a small controlled power supply would be a good feature, and not hard to do.
> * Enough cpu power to host an embedded webpage instead of just a serial console by pretending to be a usb network gadget with a local mdns url (e.g. buspirate.local )? (Thought this may be a bit of a wishful thinking).

> * (Is this even practical or sensible?) a small ICE40 frontend so you can fully configure the io pin anyway you want?
I was thinking about moving from the header to included probes with better connector, and using a digital matrix switch to implement that, but they are quite pricey. ICE40 are actually cheaper, so not that crazy of an idea.

mofosyne, 2020-04-08:
The VPU to VCC is basically about saving a single wire for the instance of quickly hooking up to an i2c module. 

Right now doing so take a bit more time scrounging for either a breadboard or soldering a wire if the module only has one VCC input.

ddavidebor, 2020-04-08:
allright got it

tomtastic, 2020-06-02:
Everything I'd like to see is being incorporated into Glasgow : https://github.com/GlasgowEmbedded/glasgow

ddavidebor, 2020-06-02:
Interesting project. Hardware seems really clever. Software too clever. I
don't like the software dependencies, nor the fact that a lot of important
protocols are missing, while there are a lot of niche/hobbyist ones
implemented.

tomtastic, 2020-06-03:
Which useful protocols are missing in your opinion?

- 1-Wire - Planned
- I2C - Yes
- SPI - Yes
- JTAG - Yes
- Asynchronous serial - Yes
- MIDI - Possible
- PC keyboard - Possible
- HD44780 LCD - Possible

As far as not liking the fact it uses the ubiquitous Python3 language to program the FPGA, I'd say anyone that has tried to compile a BusPirate firmware with MPLAB and XC-16 would probably prefer the Glasgow.

ddavidebor, 2020-06-03:
1-wire mostly. 

PICs are horrible to program, this has never been news. 

But Glasgow requires a working compiler to be used. This means that you can't throw it into the toolbox, forget about it for a year and expect it to work. There is going to be updating and maintenance to do on the computer. So the design fails to meet the criteria for it to be a tool.

This is without considering the fact that programming requires quite a few working binaries. chance of stuff breaking increases much faster than linear with the number of binaries. Those binaries are going to be a huge shitshow to run on Windows.

And whoever had the idea to program it with their own python-based hardware design language, it has doomed it to the same life as every project where someone decided to both implement a functional goal AND recreate the wheel. That is, they never go as fast as one would have tough, lag behind the competition, then people get bored of wasting their time with no results and everything gets abandoned.

To my consideration, that's not a tool, that's a well designed demoboard.

tomtastic, 2020-06-03:
I think the BusPirate development was pretty doomed too, have you counted the outstanding issues with the community firmware, almost nothing works reliably and is trial and error which firmware works with what :)
Anyway, I'm not here to bash the BusPirate, I've enjoyed using mine many times, but I'm afraid it (even the v4) is already an ancient platform now with very little active development.

ddavidebor, 2020-06-03:
I am pointing towards the hydrabus as my next tool. In fact I'm looking
towards forking the HW in the next months to improve the pinouts and I/O
compatibility.

chepo92, 2020-07-17:
I just submitted and issue #142  before reading this

IMHO, working on a new hw would be great, but also needs a working realiable and more universal firmware first, 
In order to refloat this project, it definetly has to be ported to support and be run on modern chips, ideally should be uC agnostic, STM32/ESP32 or other state of the art and cheap 32bit boards, this would make it more universal to use, considering that nowadays many makers/hobbyst come from the Arduino world, and the more electronics-using non standard chip guys (me included) always find that needing a special programmer (STlink Jlink, or even a BusPirate board) and a license to compile that cost 2x-10x the price of the target chip, is just a bit expensive (I know the companies have to subsist, but that should be for Bussiness to Bussiness, not a weekend hobbist). 

I mean, if one could compile and run BusPirate fw in an Arduino (wish it was fast enough), everyone would use it. 

The glassgow looks great, but is hardware dependant (also very sofisticated and advanced to use an FPGA), and maybe not everyone wants to debug at GHz.

mofosyne, 2020-07-19:
>  I know the companies have to subsist, but that should be for Business to Business, not a weekend hobbyist.

I really hope there is a way to figure out a cooperative organisation structure that could provide sustainable way to grow and maintain open hardware. But that's a topic for another time perhaps.

chepo92, 2020-07-19:
> I really hope there is a way to figure out a cooperative organisation structure that could provide sustainable way to grow and maintain open hardware. But that's a topic for another time perhaps.

There are many ways, just see many projects here in github that are normally sponsored or have a patreon with many supporters, maybe they are not the most profitable bussiness but at least some can sustain or partially afford it's own continuity, but it has to be constantly maintained, well organized and with a minimum critical mass of users, also a very motivated developer team. 

If I find some time i would dedicate it to start porting BP firmware to other boards

ddavidebor, 2020-07-22:
@chepo92 please look into the hydrabus, I have researched this topic extensively and it's the best successor

chepo92, 2020-07-22:
Thanks for the tip @ddavidebor 
I also found and bought this cheap board [FT2232HL](https://es.aliexpress.com/item/32806818411.html?spm=a2g0s.9042311.0.0.45b263c0BpfIU2) to test (my objective is something good and cheap, I know it sounds like a rat, but I think many makers prefer cheap toys to play, explore, learn and break) 
As I read some articles in the [net](https://www.allaboutcircuits.com/technical-articles/getting-started-with-openocd-using-ft2232h-adapter-for-swd-debugging/)

hydrabus looks very promising, will look after it when I have more budget as a pro tool, besides, it doesn't currently ship to my country. Do you have one?

ddavidebor, 2020-07-22:
No, I don't like their HW, I'm not buying it
But i'm considering making a competing hardware board

PSLLSP, 2020-08-10:
What about AxiCat? I have no experience with that device but it seems it is similar device like BusPirate but build with AVR MCU... Price is similar to price of BusPirate.

https://www.elektor.com/axicat

TechLobo, 2022-07-08:
> No, I don't like their HW, I'm not buying it But i'm considering making a competing hardware board
> […](#)

Did you ever progress this alternative board?