ElderlyPirate/Bus_Pirate #139: Backspace echoing incorrectly

yodaldevoid, 2020-05-21:
Starting with commit 06a1f1254c1e310688ec6b4aa6e8454789d47612 the backspace handling has broken. When the backspace key is hit, rather than clearing the character before the cursor and moving the cursor back, the character before the cursor is cleared, but the cursor is not moved back.

I attempted to figure out what is going wrong here, but everything seems fine to me. Strangely, though they were also modified in that commit, the arrow keys still work without fail.

wmpluto, 2020-08-14:
Hi yodaldevoid,

Please kindly check my pull request for the solution.

tudstudent, 2026-10-22:
This was fixed by #182 (merged): the destructive-backspace string is sent correctly again, so the character is cleared and the cursor moves back. The earlier PR #149 was closed in favour of #182. Can be closed.

closed 2026-10-22 by yodaldevoid