ElderlyPirate/Bus_Pirate #148: Internal timer for timestamps is missing

PSLLSP, 2020-08-09:
BusPirate doesn't have a timer for timestamps. It could be useful to have a internal timer running all the time and value of such timer could be print by user command and read in BASIC script. Maybe add a feature to reset that timer.

I assume that command `t` will print internal timer and `T` will reset internal timer. Measure how long it takes to read sample from ADC. First try to measure delay function of 1ms, than 100ms, then single ADC and than 10 ADC samples:

`HiZ>Tt%t%:100tdtddddddddddt`

That command will end with error because `t` and `T` commands are not available in the current firmware...

I think this could be useful to measure signal timings. I have a BASIC script to search for long pulses in analog signal read by ADC. I measure pulse length with simple counter `LET C=C+1`. It works but whenever I modify my BASIC program, I read different numbers because modified program runs with different timing. I need hardware timer!

HINT: Nice part to play with BusPirate is 2-pin "smart" color LED that changes colors in a pattern. These LED diodes are not calibrated so when you have several such LEDs, they will blink in a little different way. Each color have different forward voltage and BusPirate can measure forward voltage of blinking LED with ADC. It is an easy source of variable input voltage for experiments with ADC...

Student with BP and several blinking LEDs can try:
- to find two LEDs with similar blinking frequency
- to find the LED that blinks at the highest frequency
- to find the LED that blinks at the lowest frequency
- to find minimal forward voltage
- to find maximum forward voltage
- to detect that LED is missing
- to detect LED that is not blinking (single color LED)
- to detect that LED was replaced with other blinking LED
- to test how repeatable is this measurement
- many more, so much fan ;-)
- even more fun when internal timer is available...

Simple example, read forward voltage of blinking LED in BASIC, find the lowest and highest voltage. Blinking LED is powered from AUX output with 100ohm resistor in series:
```
HiZ(BASIC)>list

10  LET M=9999
20  LET N=0
30  AUX 1
40  FOR I=1 TO 10000
50  LET A= ADC 
60  REM  PRINT I;":";A
70  IF A<M THEN  LET M=A
80  IF A>N THEN  LET N=A
90  NEXT I
100  AUX 0
110  PRINT "THE LOWEST VOLTAGE: ";M
120  PRINT "THE HIGHEST VOLTAGE: ";N
65535  END 
154 bytes.
Ready
HiZ(BASIC)>run
THE LOWEST VOLTAGE: 329
THE HIGHEST VOLTAGE: 503

Ready
HiZ(BASIC)>
```

tudstudent, 2026-10-23:
Implemented in #216.

A free-running 32-bit timer rides on timer #1, which already free-runs at
2 MHz for the delay functions: its overflow interrupt maintains the high
16 bits, and the microsecond value is built as `((counter << 16) | TMR1) >> 1`.
`t` prints it, `T` restarts it from zero, and both work in every mode. It
wraps after about 35 minutes.

Your example line now runs for real - this is a v4 with the change, nothing
connected to the probe:

```
HiZ>Tt%t%:100tdtddddddddddt
2
DELAY 1ms
1872
DELAY 100ms
102757
VOLTAGE PROBE: 0.00V
... (ten probes)
104848
HiZ>
```

- `Tt` reads `2` right after the reset,
- the 1 ms delay plus the `%:100` repeat lands at 102.8 ms (wraps included),
- and the ten single ADC probes span 2.09 ms - about 209 us per sample, one
  of the timings you wanted to measure.

Builds warning-free with XC16 2.10 for both the v3 (81% program) and the v4
(26% program).

Two parts of your post are deliberately not in this pull request: reading
the timer from BASIC (a token/statement so scripts can use it instead of the
`LET C=C+1` counter), and the ADC formatting ideas (a raw value alongside
the volts, or a `VOLTS` helper). Both look like sensible follow-ups of their
own.

tudstudent, 2026-10-23:
Follow-up to my earlier comment: both remaining items are now implemented
and verified.

* **BASIC `TIME`** - reads the timestamp timer from inside a program, in
  units of 64 us (fine enough for the ADC sample timing, wide enough that a
  single 16-bit value still covers your blinking-LED pulse measurements).
  Subtract two readings for an interval; the `LET C=C+1` loop counter can
  retire.
* **The raw value in the voltage-probe command**, exactly the "Future
  firmware ;-)" format from your post:

      HiZ>d
      VOLTAGE PROBE: 0.00V (raw 0)
      HiZ>

Both are in #219 (stacked on #216, which adds the timer itself).
Verified on hardware: `TIME` tracks wall-clock delays to within one 64 us
unit (500 ms -> 7813 units, 100 ms -> 1563 units), and the displayed voltage
matches its raw value through the same conversion as before. The `VOLTS`
helper idea from your post is not included - the raw display covers that
bullet, and a printing function would be its own design discussion.