ElderlyPirate/Bus_Pirate #163: Building project fails

pezosanta, 2022-02-03:
Hello, 

I'm brand new to the Bus Pirate universe and just wanted to build and flash the firmware based on your [docs](https://github.com/BusPirate/Bus_Pirate/blob/master/Documentation/building-and-flashing-firmware.md), but got some building errors.

Hardware: Bus Pirate v3.6a
Operating System: Windows 10

Any ideas on how to solve this?

`make -f nbproject/Makefile-BusPirate_v3.mk SUBPROJECTS= .build-conf
make[1]: Entering directory 'C:/Users/S�nta P�ter/Desktop/Evoaid/workspace/volan/Bus_Pirate/Firmware/busPirate.X'
make  -f nbproject/Makefile-BusPirate_v3.mk dist/BusPirate_v3/production/busPirate.X.production.hex
make[2]: Entering directory 'C:/Users/S�nta P�ter/Desktop/Evoaid/workspace/volan/Bus_Pirate/Firmware/busPirate.X'
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../base.c  -o build/BusPirate_v3/production/_ext/1472/base.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/base.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../1wire.c  -o build/BusPirate_v3/production/_ext/1472/1wire.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/1wire.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../basic.c  -o build/BusPirate_v3/production/_ext/1472/basic.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/basic.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../bitbang.c  -o build/BusPirate_v3/production/_ext/1472/bitbang.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/bitbang.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../dio.c  -o build/BusPirate_v3/production/_ext/1472/dio.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/dio.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../jtag.c  -o build/BusPirate_v3/production/_ext/1472/jtag.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/jtag.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../jtag/lenval.c  -o build/BusPirate_v3/production/_ext/760884769/lenval.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/760884769/lenval.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../jtag/ports.c  -o build/BusPirate_v3/production/_ext/760884769/ports.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/760884769/ports.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../jtag/micro.c  -o build/BusPirate_v3/production/_ext/760884769/micro.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/760884769/micro.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../main.c  -o build/BusPirate_v3/production/_ext/1472/main.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/main.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../pc_at_keyboard.c  -o build/BusPirate_v3/production/_ext/1472/pc_at_keyboard.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/pc_at_keyboard.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../pic.c  -o build/BusPirate_v3/production/_ext/1472/pic.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/pic.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../raw3wire.c  -o build/BusPirate_v3/production/_ext/1472/raw3wire.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/raw3wire.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../selftest.c  -o build/BusPirate_v3/production/_ext/1472/selftest.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/selftest.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../raw2wire.c  -o build/BusPirate_v3/production/_ext/1472/raw2wire.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/raw2wire.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../smps.c  -o build/BusPirate_v3/production/_ext/1472/smps.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/smps.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../sump.c  -o build/BusPirate_v3/production/_ext/1472/sump.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/sump.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../i2c.c  -o build/BusPirate_v3/production/_ext/1472/i2c.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/i2c.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"C:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../hd44780.c  -o build/BusPirate_v3/production/_ext/1472/hd44780.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/hd44780.o.d"        -g -omf=elf [... cut ...]
nbproject/Makefile-BusPirate_v3.mk:412: recipe for target 'build/BusPirate_v3/production/_ext/1472/proc_menu.o' failed
make[1]: *** [.build-conf] Error 2
make: *** [.build-impl] Error 2
make[2]: Leaving directory 'C:/Users/S�nta P�ter/Desktop/Evoaid/workspace/volan/Bus_Pirate/Firmware/busPirate.X'
nbproject/Makefile-BusPirate_v3.mk:91: recipe for target '.build-conf' failed
make[1]: Leaving directory 'C:/Users/S�nta P�ter/Desktop/Evoaid/workspace/volan/Bus_Pirate/Firmware/busPirate.X'
nbproject/Makefile-impl.mk:39: recipe for target '.build-impl' failed

BUILD FAILED (exit value 2, total time: 5s)`

peterbuxton, 2022-05-10:
I am trying to build the firmware in XPLAB X IDE v6.00 with XC16 v2.00 in Windows 10 and end up with nearly identical errors as pezosanta did, ending with BUILD FAILED.

I have no idea what needs to be done. I also don't quite understand why this firmware needs to be built. Can't there just be a pre-built file available to program into the Bus Pirate with Pickit 3?

nelsonov, 2022-05-13:
I am experience what seems to be the same problem.  Upon first opening the project I get:
```
+ Configuration "BusPirate_v4" in project "busPirate": Pack PIC24F-GA-GB_DFP 1.0.28 used in configuration is not installed and not available online. 
Please install the same version of pack or switch to latest version.
Install PIC24F-GA-GB_DFP 1.0.28
```

Then running the build generates the following output:
```
make -f nbproject/Makefile-BusPirate_v3.mk SUBPROJECTS= .build-conf
make[1]: Entering directory 'D:/src/Bus_Pirate/Firmware/busPirate.X'
make  -f nbproject/Makefile-BusPirate_v3.mk dist/BusPirate_v3/production/busPirate.X.production.hex
make[2]: Entering directory 'D:/src/Bus_Pirate/Firmware/busPirate.X'
"D:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../1wire.c  -o build/BusPirate_v3/production/_ext/1472/1wire.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/1wire.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"D:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../base.c  -o build/BusPirate_v3/production/_ext/1472/base.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/base.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"D:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../basic.c  -o build/BusPirate_v3/production/_ext/1472/basic.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/basic.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"D:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../bitbang.c  -o build/BusPirate_v3/production/_ext/1472/bitbang.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/bitbang.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"D:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../dio.c  -o build/BusPirate_v3/production/_ext/1472/dio.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/dio.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"D:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../jtag/lenval.c  -o build/BusPirate_v3/production/_ext/760884769/lenval.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/760884769/lenval.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"D:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../jtag.c  -o build/BusPirate_v3/production/_ext/1472/jtag.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/jtag.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"D:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../jtag/micro.c  -o build/BusPirate_v3/production/_ext/760884769/micro.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/760884769/micro.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"D:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../jtag/ports.c  -o build/BusPirate_v3/production/_ext/760884769/ports.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/760884769/ports.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"D:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../main.c  -o build/BusPirate_v3/production/_ext/1472/main.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/main.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
"D:\Program Files\Microchip\xc16\v2.00\bin\xc16-gcc.exe"   ../pc_at_keyboard.c  -o build/BusPirate_v3/production/_ext/1472/pc_at_keyboard.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/pc_at_keyboard.o.d"        -g -omf=elf -DXPRJ_BusPirate_v3=BusPirate_v3      -std=c99 -mlarge-code -mlarge-data -O1 -fomit-frame-pointer -I".." -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -save-temps -finline 
[... cut ...]
../proc_menu.c:1847:50: error: comparison between pointer and integer
make[2]: *** [build/BusPirate_v3/production/_ext/1472/proc_menu.o] Error 255
make[1]: *** [.build-conf] Error 2
make: *** [.build-impl] Error 2
make[2]: Leaving directory 'D:/src/Bus_Pirate/Firmware/busPirate.X'
nbproject/Makefile-BusPirate_v3.mk:91: recipe for target '.build-conf' failed
make[1]: Leaving directory 'D:/src/Bus_Pirate/Firmware/busPirate.X'
nbproject/Makefile-impl.mk:39: recipe for target '.build-impl' failed

BUILD FAILED (exit value 2, total time: 7s)
```

peterbuxton, 2022-05-13:
@nelsonov , I just recently installed and became acquainted with MPLAB X for this project and have get a message like that. I clicked the Upgrade button and it said it upgraded it. However, I get the same message every time I open the project so it doesn't seem to be actually be upgrading it. Anyway, I'm only interested in the v3 section, not v4 as the message implies. I really don't know enough about all this to figure anything out.

joh, 2022-09-19:
Please try the following patch:

```
diff --git a/Firmware/openocd.h b/Firmware/openocd.h
index 3c2db24..6a59f8a 100644
--- a/Firmware/openocd.h
+++ b/Firmware/openocd.h
@@ -20,6 +20,8 @@
 
 #include "configuration.h"
 
+#define min(X, Y) (((X) < (Y)) ? (X) : (Y))
+
 #ifdef BP_JTAG_OPENOCD_SUPPORT
 
 void binOpenOCD(void);
diff --git a/Firmware/proc_menu.c b/Firmware/proc_menu.c
index e28861e..9ef8cc5 100644
--- a/Firmware/proc_menu.c
+++ b/Firmware/proc_menu.c
@@ -1844,7 +1844,7 @@ void remove_current_character_from_command_line(void) {
     cmdbuf[index] = cmdbuf[index + 1];
 
     /* Write the moved character. */
-    user_serial_transmit_character(cmdbuf[index] != NULL ? cmdbuf[index] : ' ');
+    user_serial_transmit_character(cmdbuf[index] != 0x00 ? cmdbuf[index] : ' ');
 
     /* Update pointer. */
     characters_to_move++;
```

PR #165 (2022-12-26): Fix compile errors with XC16 v2.00
Fixes #163 

I ran into the same issue compiling the firmware for the first time. These seem like pretty straightforward fixes to me, but I could be wrong. I have very little experience with the Bus Pirate, and was unable to even confirm that I'm hitting these code paths, but the firmware compiles now and I believe these are equivalent statements.

### Compiler error:
> ../openocd.c:215:7: error: implicit declaration of function 'min'

Remove calls to `min()` since I can't find it in the headers. I suspect there _is_ a `min` function we could use for readabil [... cut]
closed 2022-12-26 by andrewd207