ElderlyPirate/Bus_Pirate #168: Release (hex file) for BPv3?

nelsonov, 2022-05-13:
Several people, including me, are having trouble compiling with several compiler errors.  There is a release available, from 2016, but only for BPv4.  Would it be possible to provide a pre-built release for the v3?

brant-ci, 2023-01-03:
I've just come from the depths of google, I've been through half a dozen forum posts trying to get things sorted out.

The given process of "download this 4GB IDE to build firmware from scratch" is bad.

The fact that I have to open the hex file in an editor to _replace all lowercase letters with uppercase_ is frankly unacceptable.

Simply distributing a binary solves compiler issues and random toolchain problems. There really is no valid reason to _not_ provide a 150kb binary

AreYouLoco, 2023-01-04:
Trust me I have build the 7.0/7.1 versions and they are unstable you want to stick with older ones.
AFAIR I am using 6.3

psolyca, 2023-01-04:
I have built last one with patches from different PR.
I also fixed a JTAG issue.
I will provide a build with all warning due to the fact I didn't test all features.