ElderlyPirate/Bus_Pirate #171: Add members to the project

matsievskiysv, 2022-09-20:
This project has not seen any updates in a while. Maybe some active contributors could be given write access so development could move on?

DerekGn, 2022-10-06:
@matsievskiysv 

Another, **less than ideal**, idea maybe to fork the project and continue the development on the fork.

You can create a forked repo and then add the BusPirate/Bus_pirate repo as an upstream repo

`git remote add upstream https://github.com/BusPirate/Bus_Pirate.git`

You can then pull all the open PRs to the forked repo

`git fetch [NAME REMOTE PROJECT] pull/[PR NUMBER]/head:[PR BRANCH NAME]`

For example pull PR 170 changes to a branch called pr_name

`git fetch upstream pull/170/head:pr_name`

That way all of the contributions can be merged into the fork.

DerekGn, 2022-10-06:
@matsievskiysv Hey no problem

psolyca, 2022-11-17:
Done https://gist.github.com/psolyca/6423302561f4cc6be46716c201758abb
Everyting is compiling but currently I did not test it on the board.

ojg78, 2023-03-19:
Is the gist going to become a full supported fork? I'm trying to work out if this project is dead\dying or if getting the v4 hardware would be a good idea.

psolyca, 2023-03-19:
Sorry I have built a bin and used it but forgot to update the gist 😅
I'm just using the UART mode.

matsievskiysv, 2023-03-20:
> I'm trying to work out if this project is dead\dying or if getting the v4 hardware would be a good idea.

IMHO this firmware is good enough for many applications. It's my go-to device for getting acquainted with the new chips.

ojg78, 2023-03-21:
Is there a consensus on choosing v4 vs V3. X? Thanks for the responses.

matsievskiysv, 2023-03-22:
> Is there a consensus on choosing v4 vs V3. X? Thanks for the responses.

I'm a simple man, I just go with higher number.
I don't remember having any v4 specific issues.