ElderlyPirate/Bus_Pirate #175: pirate-loader_lnx no such file

rana97ahmad, 2023-03-09:
Hello, 
I am trying to update the firmware on my bus pirate v3.6a (Firmware v5.10 Bootloader v4.4) and my boot loader , but when I run the pirate-loader_lnx file (attached in the image below), it says no such file or directory.
I don't know what the problem could be or how to fix it? 
[image]

EunMinSoo, 2023-03-14:
I think it's because the pirate-loader_lnx binary architecture is different from the system. Executing "build-unix.sh" will recreate pirate-loader_lnx

mikebdp2, 2025-07-10:
@rana97ahmad @EunMinSoo There has also been a recent update to pirate-loader not so long ago, please check out the latest pull request in this repository