ElderlyPirate/Bus_Pirate #176: Error loading library pic30-elf

camilo-nunez, 2023-04-01:
Hi,

I am trying to follow the section [Building](https://github.com/BusPirate/Bus_Pirate/blob/master/Documentation/building-and-flashing-firmware.md#building)  of the firmware, and in MPLAB X IDE output I got this:

```
Info: Loading file: ./../p24FJ256GB106.gld
make  -f nbproject/Makefile-BusPirate_v4.mk dist/BusPirate_v4/production/busPirate.X.production.hex
make[2]: Entering directory '/home/efernandez/Downloads/Bus_Pirate/Firmware/busPirate.X'
"/opt/microchip/xc16/v2.00/bin/xc16-gcc"   -o dist/BusPirate_v4/production/busPirate.X.production.elf  build/BusPirate_v4/production/_ext/1472/1wire.o build/BusPirate_v4/production/_ext/1472/base.o build/BusPirate_v4/production/_ext/1472/basic.o build/BusPirate_v4/production/_ext/1472/bitbang.o build/BusPirate_v4/production/_ext/1241334144/cdc.o build/BusPirate_v4/production/_ext/1472/dio.o build/BusPirate_v4/production/_ext/1472/jtag.o build/BusPirate_v4/production/_ext/760884769/lenval.o build/BusPirate_v4/production/_ext/760884769/micro.o build/BusPirate_v4/production/_ext/760884769/ports.o build/BusPirate_v4/production/_ext/1472/main.o build/BusPirate_v4/production/_ext/1472/pc_at_keyboard.o build/BusPirate_v4/production/_ext/1472/pic.o build/BusPirate_v4/production/_ext/1472/raw2wire.o build/BusPirate_v4/production/_ext/1472/raw3wire.o build/BusPirate_v4/production/_ext/1472/selftest.o build/BusPirate_v4/production/_ext/1472/smps.o build/BusPirate_v4/production/_ext/1472/sump.o build/BusPirate_v4/production/_ext/1241334144/usb_stack.o build/BusPirate_v4/production/_ext/1472/onboard_eeprom.o build/BusPirate_v4/production/_ext/1472/i2c.o build/BusPirate_v4/production/_ext/1472/hd44780.o build/BusPirate_v4/production/_ext/1472/spi.o build/BusPirate_v4/production/_ext/1472/uart.o build/BusPirate_v4/production/_ext/1472/openocd.o build/BusPirate_v4/production/_ext/1472/messages_v4.o build/BusPirate_v4/production/_ext/1472/messages.o build/BusPirate_v4/production/_ext/1472/binary_io.o build/BusPirate_v4/production/_ext/1472/proc_menu.o build/BusPirate_v4/production/_ext/1472/core.o build/BusPirate_v4/production/_ext/1472/uart2.o build/BusPirate_v4/production/_ext/1472/aux_pin.o      -mcpu=24FJ256GB106        -omf=elf -DXPRJ_BusPirate_v4=BusPirate_v4      -std=c99 -Wl,,,--defsym=__MPLAB_BUILD=1,,--script="../p24FJ256GB106.gld",--check-sections,--data-init,--pack-data,--handles,--isr,--gc-sections,--fill-upper=0,--stackguard=16,--library=pic30-elf,--library-path="../../../../Program Files/Microchip/MPLAB C30/lib",--library-path="..",--no-force-link,--smart-io,-Map="BusPirate_V4.map",--report-mem,--cref,--warn-section-align,--memorysummary,dist/BusPirate_v4/production/memoryfile.xml  -mdfp="/opt/microchip/mplabx/v6.05/packs/Microchip/PIC24F-GA-GB_DFP/1.8.217/xc16" 
/opt/microchip/xc16/v2.00/bin/bin/../bin/elf-ld: cannot find -lpic30-elf
make[2]: Leaving directory '/home/efernandez/Downloads/Bus_Pirate/Firmware/busPirate.X'
make[2]: *** [nbproject/Makefile-BusPirate_v4.mk:501: dist/BusPirate_v4/production/busPirate.X.production.hex] Error 255
make[1]: *** [nbproject/Makefile-BusPirate_v4.mk:85: .build-conf] Error 2
make[1]: Leaving directory '/home/efernandez/Downloads/Bus_Pirate/Firmware/busPirate.X'
make: *** [nbproject/Makefile-impl.mk:39: .build-impl] Error 2

BUILD FAILED (exit value 2, total time: 657ms)
```

In the `configurations.xml` file of the `BusPirate_v4` exist the property `<property key="input-libraries" value="pic30-elf"/>`, but I not sure if I have installed the `pic30-elf` library.

What can I do ?

psolyca, 2023-04-04:
It could be a bad library path.
On MPLAB IDE X you should have somewhere the ability to check and install old libraries.
I'm not on the good computer so I can't check where and how.

Topper69, 2023-04-05:
Hi,
I have removed the line in the <code>configurations.xml</code> file and it compile successfully
<code>
BUILD SUCCESSFUL (total time: 2s)
</code>

camilo-nunez, 2023-04-05:
Wonderful, your solution worked for me @Topper69, thanks !

closed 2023-04-05 by camilo-nunez