ElderlyPirate/Bus_Pirate #177: BP v4 - Unable to update firmware -- Impossible to boot in booloader mode

Topper69, 2023-04-03:
Hi,

I have just received my BP v4 and i tryed to update the firmware.
Following the guide, i try: 
1) minicom 
<code>
HiZ>$
Are you sure? y
BOOTLOADER
</code>
then the board reboots and minicom reconnect with the pompt. "HiZ>" like nothing happen.

2) By shorting PGC & PGD before connecting the boad: it have the same prompt has usual, so defenitly not in bootloader mode.

3) Both method (by tiping "$" then Y with the short on PGC&PCD), same problem.

Try on linux (rasp4) and win10, same issue impossible to get the boad in the bootloader mode.
Of course the pirate-loader fails.
<code>
udo ./pirate-loader_lnx --dev=/dev/ttyACM0 --hello
+++++++++++++++++++++++++++++++++++++++++++
  Pirate-Loader for BP with Bootloader v4+  
  Loader version: 1.0.2  OS: Linux
+++++++++++++++++++++++++++++++++++++++++++

Opening serial device /dev/ttyACM0...OK
Configuring serial port settings...OK
Sending Hello to the Bootloader...ERROR
No reply from the bootloader, or invalid reply received: 0
Please make sure that PGND and PGC are connected, replug the device and try again
</code>

My board : 
<code>
HiZ>i
Bus Pirate v4
Community Firmware v7.0 - goo.gl/gCzQnW 
DEVID:0x1019 REVID:0x0004 (24FJ256GB106 UNK)
http://dangerousprototypes.com
</code>
dmeg logs
</code>
[603211.952708] usb 1-1.1: new full-speed USB device number 86 using xhci_hcd
[603212.069314] usb 1-1.1: New USB device found, idVendor=04d8, idProduct=fb00, bcdDevice= 0.02
[603212.069334] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[603212.069340] usb 1-1.1: Product: Bus Pirate V4 
[603212.069345] usb 1-1.1: Manufacturer: Dangerous Prototypes
[603212.069350] usb 1-1.1: SerialNumber: 00000001
[603212.074778] cdc_acm 1-1.1:1.0: ttyACM0: USB ACM device
</code>

Any help?

Topper69, 2023-04-04:
I have already tried shorting PGD & PGC and no success. My buspirate stay in "normal" mode.
But i'll try between <code>GND</code> and <code>PGC</code>

For now it's run in v7.0 firwmare, so the bootloader is not too old. With this firwmare JTAG is not working with openocd.

psolyca, 2023-04-04:
For JTAG I have made a fix on my merged branch
https://github.com/psolyca/Bus_Pirate/commits/merged
I have a build but did not published it... 😭

Topper69, 2023-04-04:
Could you share a .hex file for v4 board please?
It'll be very helpful !!

Many thanks.

ojg78, 2023-05-02:
Just in case this jogs anyone's community spirit, I have this same issue and would certainly appreciate wisdom from anyone who has managed to update a fairly recently purchased bpv4. I am thinking that a pickit is going to be needed.

Topper69, 2023-05-02:
> Just in case this jogs anyone's community spirit, I have this same issue and would certainly appreciate wisdom from anyone who has managed to update a fairly recently purchased bpv4. I am thinking that a pickit is going to be needed.

Hi,
After unsuccessful tries with the USB bootloader mode, i bought a pickit to successfully flash it with the MPLAB IDE & ICSP header.