ElderlyPirate/Bus_Pirate #181: please... binary for BP v3b?

petaflot, 2024-01-21:
Hi

I've been struggling with my 3D printer for longer than I'd like to admit, running into a mass of out-of-this-world issue, the latest being that the bootloader of the STM32G0B1RE sometimes refuses to update the firmware from the SD card, for no apparent reason.

So I figured I could use my old BusPirate, but its firmware can't do JTAG.. and I just can't go through the trouble of installing this proprietary MPLAB stuff on my machine and all and all.. I don't have a windows, I don't have an Arch linux, I don't have an Ubuntu, just this Gentoo that is great for so many things  but not for this.

My energy level is at close to zero.

Can someone please provide e with a binary for a BusPirate v3b from jan 2010 ?

this would be massively appreciated, thanks.

kwtaylor, 2024-01-29:
See my reply here for the latest build I could find a binary of

https://github.com/BusPirate/Bus_Pirate/issues/168#issuecomment-1820551723

Note the buspirate JTAG is very rudimentary -- there are probably better options for debugging it.

petaflot, 2025-07-10:
My BP4 seems dead, does not even pass the self test. Got a v6 in the meantime, thank you

mikebdp2, 2025-07-10:
@petaflot A lot of years ago I had the erroneous self-test failures with the ancient bootloader/firmware, so maybe your issue is false positive and your Bus Pirate is fine - just needs a bootloader/firmware upgrade to become stable _(its preinstalled stuff really sucked to be honest)_. Please could you still try these new builds just in case they might fix your BPv4? If you have a Pickit 2 programmer by a chance, you could use an opensource pk2cmd tool to flash the whole memory with a combined HEX file to overwrite both the bootloader and firmware. Although a new pirate-loader hopefully still works too as a software means of upgrading