ElderlyPirate/Bus_Pirate #183: BPv4 Upgrade breaks USB CDC - Windows 11

htrinkle, 2024-04-05:
I have upgraded from v6,2 to the latest firmware (built using MPLAB.X, config set for PB v4, no errors).

Programmed using pirate_loader without errors.  After update the USB serial port did not come up.  Device Manager says Device Can't Start (10).  Tried reset and unplug/re-plug USB.  No luck.  Windows does detect com port and enumerates it (i.e. com5).  But port comes up in error state.

Re-programmed with v6.3-r2088 and everything works.

Happy to run further experiments if that would be helpful.

htrinkle, 2024-04-06:
Error Text: in Device Manager

> This device cannot start. (Code 10)
> 
> STATUS_DEVICE_DATA_ERROR

Events Tab:

> Device settings for USB\VID_04D8&PID_FB00\00000001 were not migrated from previous OS installation due to partial or ambiguous device match.
> 
> Last Device Instance Id: USB\VID_0483&PID_572A\00000000001A
> Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
> Location Path: PCIROOT(0)#PCI(1400)#USBROOT(0)#USB(2)
> Migration Rank: 0xF000FFFFFFFF0023
> Present: false

I have tried to uninstall the device and then unplug/re-plug, but no luck

Sodium-Hydrogen, 2024-07-11:
I've encountered this as well. I believe there is a race condition that was introduced in the USB driver changes from v6.3 to the latest firmware.

The USB light is constantly on when connected to a windows 11 machine with the `STATUS_DEVICE_DATA_ERROR`. 

Spamming the reset button will eventually allow the bus pirate to fully startup and process commands on windows 11 as normal.

I don't have very much time to look into this further, but hopefully this is a good starting point for narrowing down when this bug was introduced to the USB driver on the bus pirate v4.

Sodium-Hydrogen, 2024-07-11:
Maybe #11 is related to this issue.

mikebdp2, 2025-07-10:
@Sodium-Hydrogen @htrinkle I re-built BPv4 bootloader+firmware with the latest revision of this repository + not-merget-yet pull request improvements, and did not encounter any problem yet with the different optimization levels _(although I'm using Linux)_. At the link below you can find the latest HEX file release builds, please let me know if you have any issue even with them:
- https://github.com/BusPirate/Bus_Pirate/pull/182#issuecomment-3054409254

kallisti5, 2025-07-20:
See #191   Potentially related.  @mikebdp2 what is your kernel version?

I looked  through the linux driver for usb-serial around the 04d8 fb00 VID / PID, but don't see anything substantial changing.

kallisti5, 2025-07-20:
After reading through #11, maybe we should be building with -O0 again :-\