ElderlyPirate/Bus_Pirate #191: Linux v6.15.7 doesn't see the Bus Pirate v4 as a serial device anymore

kallisti5, 2025-07-20:
Looks like something changed with the Linux usb-serial driver? It no longer detects the Bus Pirate v4 as a serial device.

## Workaround

Force the generic usb-serial driver to see it:
```
echo "04d8 fb00" | sudo tee /sys/bus/usb-serial/drivers/generic/new_id
```

kallisti5, 2025-07-20:
Here's what the kernel produces when plugged:
```
[ 1900.427589] usb 1-3: new full-speed USB device number 32 using xhci_hcd
[ 1900.817013] usb 1-3: New USB device found, idVendor=04d8, idProduct=fb00, bcdDevice= 0.02
[ 1900.817019] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1900.817021] usb 1-3: Product: Bus Pirate V4
[ 1900.817023] usb 1-3: Manufacturer: Dangerous Prototypes
[ 1900.817024] usb 1-3: SerialNumber: 00000001
```

Here's what the kernel produces when I hack in support with the workaround:
```
[ 1900.836967] usbserial_generic 1-3:1.0: The "generic" usb-serial driver is only for testing and one-off prototypes.
[ 1900.836970] usbserial_generic 1-3:1.0: Tell linux-usb@vger.kernel.org to add your device to a proper driver.
[ 1900.836972] usbserial_generic 1-3:1.0: device has no bulk endpoints
[ 1900.837780] usbserial_generic 1-3:1.1: The "generic" usb-serial driver is only for testing and one-off prototypes.
[ 1900.837782] usbserial_generic 1-3:1.1: Tell linux-usb@vger.kernel.org to add your device to a proper driver.
[ 1900.837783] usbserial_generic 1-3:1.1: generic converter detected
[ 1900.837837] usb 1-3: generic converter now attached to ttyUSB0
```

mikebdp2, 2025-07-23:
@kallisti5 Here's my output on Linux v6.14.11 kernel of [Artix Linux](https://artixlinux.org/) _(Arch without SystemD)_ . My output is slightly different, i.e. BPv4 attaches to ttyACM0 instead of ttyUSB0
```
usb 7-2: new full-speed USB device number 2 using ohci-pci
usb 7-2: New USB device found, idVendor=04d8, idProduct=fb00, bcdDevice= 0.02
usb 7-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 7-2: Product: Bus Pirate V4 
usb 7-2: Manufacturer: Dangerous Prototypes
usb 7-2: SerialNumber: 00000001
cdc_acm 7-2:1.0: ttyACM0: USB ACM device
usbcore: registered new interface driver cdc_acm
cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
```