ElderlyPirate/Bus_Pirate #192: Building firmware for Bus Pirate v3 fails when LCD HD44780 support is enabled in configuration.h.

USBEprom, 2025-07-20:
For test purposes I am trying to build custom firmwares for the **Pirate Pirate v3** starting from the current **Bus Pirate master repository v8.0**, but I am having problems with the **LCD HD44780** support.

Into **configuration.h** I changed lines from **316** to **331** in order to enable all the features available, **BASIC** too.

#ifdef BUSPIRATEV3
#define BP_ENABLE_1WIRE_SUPPORT
#define BP_ENABLE_BASIC_SUPPORT
#define BP_ENABLE_DIO_SUPPORT
**#define BP_ENABLE_HD44780_SUPPORT**
#define BP_ENABLE_I2C_SUPPORT
#define BP_ENABLE_JTAG_SUPPORT
#define BP_ENABLE_PIC_SUPPORT
#define BP_ENABLE_PC_AT_KEYBOARD_SUPPORT
#define BP_ENABLE_RAW_2WIRE_SUPPORT
#define BP_ENABLE_RAW_3WIRE_SUPPORT
#undef BP_ENABLE_SMPS_SUPPORT
#define BP_ENABLE_SPI_SUPPORT
#define BP_ENABLE_SUMP_SUPPORT
#define BP_ENABLE_UART_SUPPORT
#endif /* BUSPIRATEV3 */

Then I changed the line **425** too in this way in order to make room for **BASIC** scripting:

**#define BP_BASIC_PROGRAM_SPACE 512**

Finally I had to set line **571** for **up 256 characters** into the command line by changing it in this way:

/**
* How big the serial command buffer can be, in bytes.
*
* @warning This must be set to a power of two, ie. 256, 128, 64, 32, etc.
*/
#ifdef BUSPIRATEV3
**#define BP_COMMAND_BUFFER_SIZE 256**
#else
#define BP_COMMAND_BUFFER_SIZE 256
#endif /* BUSPIRATEV3 */

In line **586** of **configuration.h** delay code provided is currently enabled:

**#define BP_USE_HARDWARE_DELAY_TIMER** 

I also changed **~400kHz** to **~220kHz** in line **1457** of **messages_v3.s** in order to match the real speeds now used for I2C SOFTWARE, 2-WIRE and 3-WIRE (**I have to check this**).

I am building from the source with **MPLAB-X IDE v5.00** and the **XC-16 v1.35** compiler on a **Ubuntu 18.04** computer, using the **compression option "1"** of **MPLAB** in order to be in fully accordance with the recommendations issued in order to build firmware for the **Bus Pirate v3** and v4.
By building with configurations and options I wrote, **MPLAB-X IDE v5.00** gives me the following error message and compiling fails:

_**...
"/opt/microchip/xc16/v1.35/bin/xc16-gcc"   ../hd44780.c  -o build/BusPirate_v3/production/_ext/1472/hd44780.o  -c -mcpu=24FJ64GA002  -MMD -MF "build/BusPirate_v3/production/_ext/1472/hd44780.o.d"        -g -omf=elf -save-temps=obj -DXPRJ_BusPirate_v3=BusPirate_v3  -no-legacy-libc    -std=gnu99 -msmall-code -msmall-data -msmall-scalar -mconst-in-code -O1 -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off   -finline
nbproject/Makefile-BusPirate_v3.mk:421: recipe for target 'build/BusPirate_v3/production/_ext/1472/hd44780.o' failed
make[2]: Leaving directory '/home/ubuntu/Desktop/Linux/Firmware8/busPirate.X'
nbproject/Makefile-BusPirate_v3.mk:90: recipe for target '.build-conf' failed
make[1]: Leaving directory '/home/ubuntu/Desktop/Linux/Firmware8/busPirate.X'
nbproject/Makefile-impl.mk:39: recipe for target '.build-impl' failed
elf-cc1: warnings being treated as errors
../hd44780.c: In function 'HD44780_SPIwrite':
../hd44780.c:367:9: error: implicit declaration of function 'spi_write_byte'
make[2]: *** [build/BusPirate_v3/production/_ext/1472/hd44780.o] Error 255
make[1]: *** [.build-conf] Error 2
make: *** [.build-impl] Error 2**_

_**BUILD FAILED (exit value 2, total time: 9s)**_

Please I need also clarification about the compiler **xc16-gcc** (**XC16 v1.35**) optimization options, if possible.

1) Is **Optimization level 1** still the better choice nowadays?

2) How should the item >**Do not override 'inline'**< be set?
I have not used **MPLAB-X IDE v5.00** for a long time, but if I am not mistaken, seems to me the option >**Do not override 'inline'**< had to be enabled with a check mark, now the related tag is grayed out with inside it the message **(N/A since v1.34)**.
This thing is weird, because I am pretty sure >**Do not override 'inline'**< had to be enabled with a check mark and the **xc16-gcc** compiler version on my **Ubuntu 18.04** computer it has always been **XC16 v1.35**.
Maybe **MPLAB-X IDE v5.00** I am using does it have any problems?

Be seeing you.

U.Sb

USBEprom, 2025-07-21:
OK guys.
About my second question I investigated and found the following.
In the **configurations.xml** file inside the **busPirate.X** project (**...PATH\Firmware\busPirate.X\nbproject\configurations.xml**) the revision of the **XC16** compiler assigned to the toolchain is specified both for the Bus Pirate v3, and for the Bus Pirate v4 (could be different from each other) and in the **current Bus Pirate master repository v8.0** the revision of the **XC16** compiler declared is **v2.10** (both for BPv3, and BPv4).
Although the revision of the **XC16** compiler installed on my **Ubuntu 18.04 computer** is **XC16 v1.35** and does not match the requirements into **configurations.xml** file, **MPLAB** still tries to set the possible in agreement with it and this explains why the >**Do not override 'inline'**< option becomes unavailable with the related tag grayed out with inside it the message **(N/A since v1.34)**.
The >**Do not override 'inline'**< option becomes available again by changing **configurations.xml** by imposing for the toolchain a version of the compiler **XC16** lower than **v1.34** , for instance **XC16 v1.33**.
So the question now is, to build from the source the current **firmware v8.0** for the **Bus Pirate v3**, can I change **configurations.xml** by setting into the toolchain the **XC16 v1.33** compiler (**I can not use XC16 v1.35 because the >Do not override 'inline'< option becomes unavailable**) and use **MPLAB-X IDE V5.00** with **XC16 V1.35** compiler on my **Ubuntu 18.04 computer**, or I have to honor **configurations.xml** and then upgrade my **MPLAB** enviroment by install on my **Ubuntu 18.04 computer** the current **XC16 v2.10** compiler declared for the **busPirate.X** project into its **configurations.xml**?

Another problem that I have is that I seem to understand that the current **XC16 v2.10** compiler is distributed only in the **64  bit** version and my **Ubuntu computer is 32 bit** ...
But this is another story and in any case I have to deepen the matter.

Be seeing you.

U.Sb

USBEprom, 2025-07-25:
OK guys.
Here are some news.
Now I have the latest **MPLAB-X IDE V6.25** and **XC16 V2.10** on a **Lubuntu 25.04 64 bit computer** and the environment behaves correctly with the optimization options, including >**Do not override 'inline'**< too, so my second question found the right answer, but sadly the first one do not...
In fact even now that I am building from the source with **MPLAB-X IDE v6.25** and the **XC-16 v2.10** compiler on a **Lubuntu 25.04** computer, using in **MPLAB** the **compression option "1"** and >**Do not override 'inline'**<, by building with configurations and options I wrote in my first post, **MPLAB-X IDE v6.25** still gives me the following error message and compiling fails:

**_"/opt/microchip/xc16/v2.10/bin/xc16-gcc"   ../hd44780.c  -o build/BusPirate_v3/production/_ext/1472/hd44780.o  -c -mcpu=24FJ64GA002  -MP -MMD -MF "build/BusPirate_v3/production/_ext/1472/hd44780.o.d"        -g -omf=elf -save-temps=obj -DXPRJ_BusPirate_v3=BusPirate_v3      -std=gnu99 -msmall-code -msmall-data -msmall-scalar -mconst-in-code -O1 -mcci -msmart-io=1 -Werror -Wall -msfr-warn=off -finline  -finline  -mdfp="/opt/microchip/mplabx/v6.25/packs/Microchip/PIC24F-GA-GB_DFP/1.10.375/xc16"_**
**_make[2]: Leaving directory '/home/lubuntu/Firmware8/busPirate.X'
make[1]: Leaving directory '/home/lubuntu/Firmware8/busPirate.X'
elf-cc1: warnings being treated as errors_**
**_../hd44780.c: In function 'HD44780_SPIwrite':
../hd44780.c:367:9: error: implicit declaration of function 'spi_write_byte'_**
**_make[2]: *** [nbproject/Makefile-BusPirate_v3.mk:373: build/BusPirate_v3/production/_ext/1472/hd44780.o] Error 255_**
**_make[1]: *** [nbproject/Makefile-BusPirate_v3.mk:85: .build-conf] Error 2
make: *** [nbproject/Makefile-impl.mk:39: .build-impl] Error 2_**

**_BUILD FAILED (exit value 2, total time: 4s)_**

Be seeing you.

U.Sb

denis2342, 2025-07-28:
#194 should fix this

USBEprom, 2025-07-28:
**@denis2342**

Thank you very much Sir!
I confirm the issue is gone.
Your terrific code fix the matter Sir, thanks a lot!

Be seeing you.

U.Sb